AstroPrint/AstroBox #59: Internet setup causes an Internal Server Error

MiniguyBrendan, 2015-04-29:
When I try to set up internet, this happens:

[image]

If I skip it, nothing functions properly.

Thoughts?

CoDanny, 2015-04-29:
Unfortunately we don't currently support the AstroBox project on Windows Machines. Linux or Mac are supported.

MiniguyBrendan, 2015-04-29:
Wait, you mean you can't set up the astrobox from a windows machine? That would have been nice to know when I reset my raspberry pi for it...

CoDanny, 2015-04-29:
No, I had assumed that you're fiddling with the open source project as you entered your ticket here. We do support setting up and accessing a box from Windows. Product related support tickets should go here. https://astroprint.zendesk.com

Please enter a ticket there and we will help you get you astrobox working.

closed 2015-04-29 by CoDanny

MiniguyBrendan, 2015-04-29:
Okay, one more thing, it's not an AstroBox from you guys, it's my Raspberry Pi. Thanks for the help!

MiniguyBrendan, 2015-04-29:
OK, I just read about the WiFi dongles. I'm assuming mine must not be supported. The RTL8192CU does seems to have concurrent mode, but it's not working once I get to that step.

CoDanny, 2015-04-29:
We support RPis just the same :) It would be best if we move this conversation to our support system. Could you open a ticket there?

MiniguyBrendan, 2015-04-29:
Sure, sorry about that! I'm off! Last question, will the responses be as quick as they were here?

CoDanny, 2015-04-29:
As quick and by the same person ;)

MiniguyBrendan, 2015-04-29:
Awesome, thanks! Good to know!