AstroPrint/AstroBox #60: Disable gcode analyse for x3g uploads

evils, 2015-05-05:
Uploading x3g files to Astrobox takes ages as Astrobox stops at 95% to analyse the file just uploaded. This is the correct behaviour for gcode files, though not for x3g files as Astrobox doesn't get any information out from x3g files, therefore this step is unnecessary and wastes a lot of time uploading files.

CoDanny, 2015-07-27:
We do actually get the same information from x3g files. The analysis of the file should be done on the background and should not affect the upload. Is that not the case with 0.8(0)?

evils, 2015-07-27:
It's still the case with 0.8.0. Uploads till 95% and then a long analysis starts. 
Are you sure you're getting any data from x3g files? Cause the metadata file has no information about the file in it after the analysis step and estimated print time only gets populated after the first print.

CoDanny, 2015-07-27:
I should be getting but there's most likely a bug here.

andrewcaustralia, 2015-08-03:
I see the same thing; my x3g files are created in simplify3D and uploaded to the astrobox, but it sits there on "analysing" for all x3g files and doesn't know how many layers there are, time to complete etc. 

Once it's finished a print it updates the completion time but all new x3g files just sit there.

CoDanny, 2015-08-03:
Would it be possible to post a link to one of those files so I can download and try?

evils, 2015-08-03:
Literally any x3g file works.

Here's one
https://dl.dropboxusercontent.com/u/1522254/4LandingGear.x3g

CoDanny, 2015-08-05:
I really can't make this happen. The analyzing process does nothing for x3g but I added it to the framework for future work. If you look at this portion of the code:

https://github.com/AstroPrint/AstroBox/blob/master/src/astroprint/printfiles/x3g.py#L36

you'll see how the start and finish is actually immediate. It's possible that the finished event is getting missed somehow? I can't make it happen on my environment.

We separated the analyzing part from the uploading in a previous release. Do you see the analyzing step during upload progress on top or as a status message next to the file on the list below?

Can you send some logs? they're located at /var/log/astrobox/astrobox.log

andrewcaustralia, 2015-08-10:
There's no detail in the logs regarding file uploads / analysis; does this need to be enabled somewhere else?

IE:
```
2015-06-03 01:38:49,426 - astroprint.printer - INFO - Changing monitoring state from 'Printing' to 'Operational'
2015-06-03 02:18:51,371 - octoprint.server.util - INFO - Client connection closed
2015-06-03 02:21:33,950 - octoprint.server.util - INFO - New connection from client: 10.0.0.100
2015-06-03 02:21:39,954 - astroprint.printer - INFO - Changing monitoring state from 'Operational' to 'Printing'
2015-06-03 02:21:43,795 - octoprint.server.util - INFO - Client connection closed
2015-06-03 02:26:58,515 - astroprint.printer.s3g.printjob - WARNING - Build Cancel detected
2015-06-03 02:26:59,276 - astroprint.printer - INFO - Changing monitoring state from 'Printing' to 'Operational'
2015-06-03 02:29:50,968 - octoprint.server.util - INFO - New connection from client: 10.0.0.100
2015-06-03 02:30:13,162 - astroprint.printer - INFO - Changing monitoring state from 'Operational' to 'Printing'
2015-06-03 02:30:26,946 - octoprint.server.util - INFO - Client connection closed
```

CoDanny, 2017-04-20:
Unable to reproduce. Please try again with latest version and report back if it still happens

closed 2017-04-20 by CoDanny