AstroPrint/AstroBox #62: Suggestion: Add an advanced option for inputting raw GCode

MiniguyBrendan, 2015-05-06:
I was thinking this should be placed under the control section.

I might make a Pull request if CoDanny doesn't do it :smiley:

CoDanny, 2015-05-06:
@MiniguyBrendan Go for it ! ;)

MiniguyBrendan, 2015-05-09:
Great! I'm getting there, I just need to know how to make the textbox look pretty. I see in settings.jinja2 that there's

```
							<div class="input-wrapper">
								<label for="max-nozzle-temp" class="bold">
									Max. Nozzle Temperature
								</label>
								<div class="row collapse">
							        <div class="small-9 columns">
							        	<input id="max-nozzle-temp" name="max_nozzle_temp" type="text" value="<%= settings.max_nozzle_temp %>" required pattern="integer"/>
							        	<small class="error">Invalid Nozzle Temperature.</small>
							        </div>
							        <div class="small-3 columns"><span class="postfix">&deg;C</div>
							    </div>
							</div>
```

I want a similar looking text box in control.jinja2. Are the div classes universal? Meaning, can I just copy the class name to make it look the same?

Thanks in advance,
Miniguy

CoDanny, 2015-05-10:
Yes, they are. For more info on the UI library that we use check out: http://foundation.zurb.com/docs/

MiniguyBrendan, 2015-05-10:
[image]

How's this looking so far?

CoDanny, 2015-05-10:
I think this functionality is pretty advanced so we probably would hide under an advanced option of some sort. Then I think that the user should also see that the printer responds to when the command is sent. So maybe a little history of commands received and then a box to send.

It's ok if you'd like to focus on making it work for your pull request, @joshwhite47 can  make it fit in the UI

MiniguyBrendan, 2015-05-10:
Sure! Any suggestion as to where I should move it?

joshwhite47, 2015-05-10:
Yes I can make it fit in the UI.

MiniguyBrendan, 2015-05-10:
Great. I'll focus on making it actually function for now.

MiniguyBrendan, 2015-05-10:
@CoDanny , how can I test the py files on my astrobox? They're all compiled to pyo.

MiniguyBrendan, 2015-05-10:
Oof, where is control.js located in the AstroBox? It's not under astroprint-->static-->js-->app-->views.

CoDanny, 2015-05-11:
You should develop on your mac (windows not supported for development) or linux. Connect your printer to your computer's usb port and the sofware should detect it. The final image is compiled py, js and scss files. You should not work on there.

If you must work on the box clone the repo on the box and start it there. You can start the default astrobox service like so:

sudo service astrobox stop

MiniguyBrendan, 2015-05-11:
Alright, thanks. I cloned the repo on my box, and it still searches for the .css file rather than the scss file. Any ideas?

CoDanny, 2015-05-11:
run:

sass --watch src/astrobox/static/css/scss:src/astrobox/static/css/gen &

MiniguyBrendan, 2015-05-11:
I had to run sass --watch src/astroprint/static/css/scss:src/astroprint/static/css/gen &

for it to work.  Now it says:  >>> Sass is watching for changes. Press Ctrl-C to stop.

CoDanny, 2015-05-11:
That's fine

MiniguyBrendan, 2015-05-11:
But it continues to say that, the command doesn't complete. It just stays there.

CoDanny, 2015-05-11:
Did you add the '&' at the end?

MiniguyBrendan, 2015-05-11:
Yep.

CoDanny, 2015-05-11:
Then hit enter.

It'll take a while to run the scss compiler since it's a pi but it will
eventually tell you the new css files are created

MiniguyBrendan, 2015-05-11:
I hit enter. I guess I'm just impatient haha

MiniguyBrendan, 2015-05-12:
It's also looking for app.js and other files that aren't under the gen folder. How should I get those? I can't just copy/paste from the astrobox distribution because they don't include my customized js.

CoDanny, 2015-05-13:
Sorry about the lack of instructions here. Add a file called <i>application.cfg</i> to you local directory. Use this as its contents:

    ASSETS_DEBUG = True
    ASSETS_AUTO_BUILD = True

CoDanny, 2015-08-07:
This is now part of the 0.8.1 release

closed 2015-08-07 by CoDanny