AstroPrint/AstroBox #67: Temperature resets itself

evils, 2015-05-29:
This is possibly a bug that's only happening on my MakerBot Replicator 2, though it's quite likely that this is happening on all supported printers.

What's happening is that I log into astroprint.com -> Monitor. I see my printer printing, though the Temperatur is shown as currently "--" and the set Temperatur is at zero. 
Without touch anything (except the picture button), my temperature gets set to 0 and the hot end starts cooling down. Any try (tried 20-30 times) from here on setting the temp to 230C again, doesn't work as the set temp goes down to 0 again after 1-2 seconds.
Really frustrating, as those prints die as you can imagine.

CoDanny, 2015-05-29:
How are you accessing astroprint.com? touch or non touch device?

evils, 2015-05-29:
Non touch, MBPr 15" OS X 10.10.3, latest Chrome

CoDanny, 2015-05-29:
Same setup here. Except a FlashForge Creator Pro instead of the MB. I can't see the same problem.

'--' means we're still waiting from a reported temperature from the printer. If the black handle has 0, that means the printer has set the target to 0. Something is setting it to 0. Though everything is possible, I can't currently think of a case where something in that screen would do that without user interaction.

evils, 2015-05-29:
@CoDanny 
I understand how the temperature bar works and how it's intended to work. I'm a dev myself and understand the mechanics that it's waiting for a response from the astrobox on the termperature. What makes me curios is the part where I explicitly set the Temperature back to 230C and it just goes back to 0 after a one or two seconds. Just as it would be a overwrite from the astrobox. 

I'll make you a video and a dump of the local copy once this happens again.

CoDanny, 2015-05-29:
Thanks. Also if you can enable serial logs on your astrobox and send me the logs, would help.

When you set the temp to 230, we send a command to the printer but the printer can't reject the command and leave the temp. 

Silly question but have you tried other host software with mac usb direct connection lately?

evils, 2015-05-29:
"Silly question but have you tried other host software with mac usb direct connection lately?"
What exactly do you mean?

My astrobox is a raspberry pi 1 (Model B). Otherwise I'm using the SD Card and/or the Makerbot Desktop software with my Mac

CoDanny, 2015-05-29:
I'm trying to see if there's a HW problem on your machine. Does this happen with the desktop software?

evils, 2015-05-29:
I see :)

So here's one thing that should might be taken into consideration. I'm using the latest sailfish fw on my printer. If I set the "Overwrite temperature" option on the sailfish fw, the temperature bar basically has no power over the printers temperature control and it sticks with my preheat temperature. 

I honestly can't tell you if that happens with the desktop software as I'm not printing too often with it.

evils, 2015-07-10:
@CoDanny Just had the same problem again. I was monitoring my print. Printer was about to heat up to 230 C, I didn't even remotely touch the temperature setter ,stayed at "--" for a bit and then set itself to 0. Temperature went slowly down as the printer received the 0 C.
I just send the serial logs to astroprint with this ticket in the comment section. Let me know what else I can get you

CoDanny, 2015-07-10:
Here's the relevant portion of the log with some notes.

```
2015-07-10 12:26:17,079 - SERIAL - INFO - {"event":"packet_sent", "data": "0xd5 0x03 0x0a 0x00 0x20 0x49"}   <---- This is a request to get the tool head target temp
2015-07-10 12:26:17,094 - SERIAL - INFO - {"event":"response_received", "data": "0x81 0xe6 0x00"}  <--- The response is 230 C, This is the last time 230 C is reported

2015-07-10 12:26:17,119 - SERIAL - INFO - {"event":"packet_sent", "data": "0xd5 0x03 0x0a 0x00 0x1e 0xe8"}
2015-07-10 12:26:17,155 - SERIAL - INFO - {"event":"response_received", "data": "0x81 0x00 0x00"}
2015-07-10 12:26:17,168 - SERIAL - INFO - {"event":"packet_sent", "data": "0xd5 0x03 0x0a 0x00 0x21 0x17"}
2015-07-10 12:26:17,185 - SERIAL - INFO - {"event":"response_received", "data": "0x81 0x00 0x00"}

2015-07-10 12:26:17,513 - SERIAL - INFO - {"event":"packet_sent", "data": "0xd5 0x01 0x11 0xc3"}  <-- Reset command sent
2015-07-10 12:26:17,534 - SERIAL - INFO - {"event":"response_received", "data": "0x81"} <--- Acknowledged and likely executed

Below is build start sequence

2015-07-10 12:26:17,547 - SERIAL - INFO - {"event":"packet_sent", "data": "0xd5 0x15 0x99 0x00 0x00 0x00 0x00 0x53 0x74 0x72 0x6f 0x6e 0x67 0x65 0x73 0x74 0x46 0x65 0x6d 0x61 0x6c 0x65 0x00 0xfa"}
2015-07-10 12:26:17,875 - SERIAL - INFO - {"event":"response_received", "data": "0x8c"}
2015-07-10 12:26:17,879 - SERIAL - DEBUG - {"event":"transmission_problem", "exception":"<class 'makerbot_driver.errors.PacketTimeoutError'>", "message":"", "retry_count"=0}
2015-07-10 12:26:17,885 - SERIAL - INFO - {"event":"packet_sent", "data": "0xd5 0x15 0x99 0x00 0x00 0x00 0x00 0x53 0x74 0x72 0x6f 0x6e 0x67 0x65 0x73 0x74 0x46 0x65 0x6d 0x61 0x6c 0x65 0x00 0xfa"}
2015-07-10 12:26:17,892 - SERIAL - INFO - {"event":"response_received", "data": "0x81"}
2015-07-10 12:26:17,899 - SERIAL - INFO - {"event":"packet_sent", "data": "0xd5 0x03 0x0a 0x00 0x02 0xd6"} <--- request to read current head Temp
2015-07-10 12:26:17,911 - SERIAL - INFO - {"event":"response_received", "data": "0x81 0x00 0x00"} <-- The response is 0 here but later it goes back to 230 and gradually decreases
2015-07-10 12:26:17,918 - SERIAL - INFO - {"event":"packet_sent", "data": "0xd5 0x03 0x96 0x00 0x00 0xf9"}
2015-07-10 12:26:17,930 - SERIAL - INFO - {"event":"response_received", "data": "0x81"}

2015-07-10 12:26:17,936 - SERIAL - INFO - {"event":"packet_sent", "data": "0xd5 0x03 0x0a 0x00 0x20 0x49"}  <--- This is another request to get temp target toolhead
2015-07-10 12:26:17,947 - SERIAL - INFO - {"event":"response_received", "data": "0x81 0x00 0x00"} <--- The response is 0 C
```

In summary: The printer starts reporting 230 C as the tool temp. This might be preheating as the build actually starts after. When the build starts the printer reports 0 C and by itself sets the target to 0. There's no command sent to the printer to set the temp to 0. I'm assuming that this is caused by the reset command that is sent at the start of every build.

How are you slicing your files? What seems to be missing is a command in the build instructions that sets the proper temp.

evils, 2015-07-10:
I'm using Makerbot Desktop as that'll give me the best printing results.

evils, 2015-07-10:
So after restarting the print everything works fine! So I don't see why my x3g file is causing the problem?! Maybe it's the sailfish FW

CoDanny, 2015-07-10:
Let's move this to our support ticket system at https://astroprint.zendesk.com for now. If we determine that it's a bug in the AstroBox system. We will re-open and update this ticket.

For now during troubleshooting, we should do that offline. Sound good?

closed 2015-07-10 by CoDanny

evils, 2015-08-13:
Just FYI, with today's 0.8.1. I constantly have this error no matter what. I can't change the temperature at all