AstroPrint/AstroBox #71: Custom filament temperature not being respected (Lulzbot Mini)

AdamPan, 2015-09-22:
Creating a custom filament (in my case Colorfabb XT) and putting a temperature of 240C doesn't seem to stick when I slice. Looking at the Gcode, I get the following preheat command

    M109 S205    ; set extruder temp and wait

despite setting filament temperature to 240

AdamPan, 2015-09-22:
Sample Gcode snippet included

    ;Generated with Cura_SteamEngine 15.04
    ;This Gcode has been generated specifically for the LulzBot Mini
    ;Basic settings: Layer height: 0.25 Walls: 1 Fill: 20
    ;Filament Diameter: 2.85
    ;Nozzle Size: 0.5
    G21                          ; metric values
    G90                          ; absolute positioning
    M82                          ; set extruder to absolute mode
    M107                         ; start with the fan off
    G92 E0                       ; set extruder position to 0
    M140 S60; get bed heating up
    G28                          ; home all
    M109 S140                    ; set to cleaning temp and wait
    G1 Z150 E-30 F75             ; suck up XXmm of filament
    M109 S140                    ; heat up rest of way
    G1 X45 Y174 F11520           ; move behind scraper
    G1 Z0  F1200                 ; CRITICAL: set Z to height of top of scraper
    G1 X45 Y174 Z-.5 F4000       ; wiping ; plunge into wipe pad
    G1 X55 Y172 Z-.5 F4000       ; wiping
    G1 X45 Y174 Z0 F4000         ; wiping
    G1 X55 Y172 F4000            ; wiping
    G1 X45 Y174 F4000            ; wiping
    G1 X55 Y172 F4000            ; wiping
    G1 X45 Y174 F4000            ; wiping
    G1 X55 Y172 F4000            ; wiping
    G1 X60 Y174 F4000            ; wiping
    G1 X80 Y172 F4000            ; wiping
    G1 X60 Y174 F4000            ; wiping
    G1 X80 Y172 F4000            ; wiping
    G1 X60 Y174 F4000            ; wiping
    G1 X90 Y172 F4000            ; wiping
    G1 X80 Y174 F4000            ; wiping
    G1 X100 Y172 F4000           ; wiping
    G1 X80 Y174 F4000            ; wiping
    G1 X100 Y172 F4000           ; wiping
    G1 X80 Y174 F4000            ; wiping
    G1 X100 Y172 F4000           ; wiping
    G1 X110 Y174 F4000           ; wiping
    G1 X100 Y172 F4000           ; wiping
    G1 X110 Y174 F4000           ; wiping
    G1 X100 Y172 F4000           ; wiping
    G1 X110 Y174 F4000           ; wiping
    G1 X115 Y172 Z-0.5 F1000     ; wipe slower and bury noz in cleanish area
    G1 Z10                       ; raise z
    G28 X0 Y0                    ; home x and y
    M109 S140                    ; set to probing temp
    M204 S300                    ; Set probing acceleration
    G29                          ; Probe
    M204 S2000                   ; Restore standard acceleration
    G1 X5 Y15 Z10 F5000          ; get out the way
    M400                         ; clear buffer
    G4 S1                        ; pause
    M109 S205    ; set extruder temp and wait
    G4 S15                       ; wait for bed to temp up
    G1 Z2 E0 F75                 ; extrude filament back into nozzle
    M140 S60; get bed temping up during first layer
    ;Layer count: 163
    ;LAYER:0
    M107
    G0 F10500 X35.355 Y60.450 Z0.425
    ;TYPE:SKIRT
    G1 F900 X69.570 Y60.449 E1.13971
    G1 X70.922 Y61.801 E1.20340

and so on

CoDanny, 2015-12-22:
Is this still an issue? There was fix in our cloud slicer that should have addressed this.

closed 2016-02-21 by CoDanny