AstroPrint/AstroBox #72: Canceling a job can cause a printhead crash

AdamPan, 2015-09-22:
On some printers, canceling a job can cause the printer to home X/Y at too low of a Z height, causing the extruder to crash into parts of the print bed. 

For example, the Lulzbot Mini has conductive autolevelling disks on the bed which the extruder will run into when homing at too low of a Z height.

A potential solution to this would be to home the Z-axis before homing the X/Y axes.

Ideally the user should be able to choose a cooling location, and the order to move each axis to avoid crashing (perhaps an advanced option?).

mobileoverlord, 2015-11-17:
I am experiencing the same issue with my Ultimaker 2. It crashes with the clip on the print bed if the print is cancelled during one of the initial layers. I think that to solve this problem, the printer should execute the GCode end commands in the printer profile instead of hard coding them.

mobileoverlord, 2015-11-17:
The issue seems to be here for marlin:
https://github.com/AstroPrint/AstroBox/blob/master/src/astroprint/printer/marlin/__init__.py#L275

I was trying to see if I could do a PR for this, but I am having trouble identifying how to get the GCode Stop from the `profileManager`

commit 43f7f147 (2015-12-22): strip commends out. Fixes #72
closed 2015-12-22 by CoDanny