AstroPrint/AstroBox #84: Increase the file-size limit of G-CODE files to 150MB when uploading to the AstroBox.

DilankaW, 2016-02-29:
Per [this thread](https://forum.astroprint.com/t/increment-gcode-upload-size/851/2), this is a request to increase the maximum file size of GCODE file(s) to 150MB when uploading to the local AstroBox.

CoDanny, 2016-03-01:
I can't find anything about a file limit for gcode uploads. I have requested more info on the error using the forums. It could be the the long file upload is being stopped because is not chunked which will be another feature to add

G33kLabs, 2016-03-01:
It seems to be a limit here is the http error :

```
Failed to load resource: the server responded with a status of 502 (Bad Gateway)
http://astrobox.local/api/files/local Failed to load resource: net::ERR_CONNECTION_RESET
```

And me too, I didn't see limit in source files.
I searched for a '502' error declaration and didn't find it in sources. 

It should be a default error emitted by the http server in python himself.

What do you think about that : http://webpy.org/cookbook/limiting_upload_size
But as I only have coded little scripts in python, I don't know which file I have to edit to add this declaration.

G33kLabs, 2016-03-01:
[image]

As you can see on the screenshot, 65Mb is not a problem for local uploads.

CoDanny, 2016-03-01:
Thanks @G33kLabs. I think it has more to do with the time it takes to upload the file than the size itself. Could you measure the time elapse between the start of your upload and the error?

G33kLabs, 2016-03-01:
Instantly

CoDanny, 2016-03-01:
Then I'm obviously wrong ;) I'll keep digging

G33kLabs, 2016-03-01:
Yes it looks really like an upload limit.
As I read that cgi is used by python, I think that the limitation can be changed and that your source use defaults which should be around 100MB.

Two of my thinks to resolve this issue :

```
Increment cgi param : 
MAX_CONTENT_LENGTH => 157286400 (150 MB)
```

```
import cgi
# Maximum input we will accept when REQUEST_METHOD is POST
# 0 ==> unlimited input
cgi.maxlen = 150 * 1024 * 1024 # 150MB 
```

But I'm really not a python developer, so perhaps I'm on a wrong way, perhaps it's the good one... :)

CoDanny, 2016-03-01:
You almost had it ;) It was a Tornado setting, the webserver we use and it was defaulting to 100MB. I have increased it to 160MB, it'll go out in the next release of the AstroBox software.

G33kLabs, 2016-03-01:
Great !!
Can you send me instructions to change it on my astrobox, I can't wait for the next release :)

CoDanny, 2016-03-01:
The above commit has the code changes but the code in your astrobox is "compiled" python... You won't have to wait long for a new release. I think this week

closed 2016-03-01 by CoDanny

G33kLabs, 2016-03-01:
Ok... Just look at the /Astrobox/src files and nope I can't edit them :)
Take care of the commented routes in your last commit and thank you for the update !

cbgj, 2017-01-16:
I am trying to upload a 234MB file, where is the setting to change to make this work?

CoDanny, 2017-01-16:
@cbgj It's currently hardcoded but I have created a task to implement this in the next release:

https://github.com/AstroPrint/AstroBox/issues/213

commit 61830b27 (2016-03-01): fixes #84