AstroPrint/AstroBox #86: Remote ON/OFF switch of the 3d printer

havanakuba, 2016-03-03:
I'm using my raspberry pi, and I came with idea to add a button to web interface to turn off your printer after finish print or printer lights during print this can be done easily on raspberry via relay and connected gpio

robermeyer, 2017-02-20:
I also need the ability to turn ON/OFF my printer(s) before and after a print.  
I have 2 delta printers so I would like to be able to turn either one ON/OFF remotely.  
Concurrently maybe turn ON/OFF a light per printer for the camera on demand.
How about an exhaust fan.
I see this is almost a year old and not assigned to anybody and no milestone.  Obviously if nobody starts on it, it will never get done.  What can be done to get this started?

havanakuba, 2017-02-20:
I was thinking to use this cheap relays.
My DIY skils are very good (due to what I do for living) but my python programming skills are not that great. But improving every day if I have some time left.

[image]

havanakuba, 2017-02-20:
You can get those as 1,2,4,8 channels, so you can control whatever you want.

havanakuba, 2017-02-20:
Some parts of Script from this video can be implemented.

robermeyer, 2017-02-20:
What video?
I agree that the hardware exists.  I was thinking SSR myself but either would work.
The SSR one is available already with a plug attached.
Just need to get it into the GUI so its useful.
In some cases you want to turn things ON/OFF with print status.
Other times it would be a simple button ON/OFF peripheral.
I would be happy with button ON/OFF remote for a start.

havanakuba, 2017-02-20:
https://www.youtube.com/watch?v=OQyntQLazMU

havanakuba, 2017-02-20:
He is using php script run from webbrowser

robermeyer, 2017-02-20:
Very nice.  Now how do we get it into AstroPrint?
Since I can't have a bunch of jumper wires laying around we would need a connector to attach to the Raspberry-Pi.

havanakuba, 2017-02-21:
I have made connection tape using dupont connector an 6 core tape
[image]

havanakuba, 2017-02-21:
I have used:
GPIO 4 as in1 
GPIO 3 as in2
GPIO 2 as in3 
GPIO 17 as in4 
[image]

havanakuba, 2017-02-21:
[image]

havanakuba, 2017-02-21:
And created 8 scripts based on the one from the video
Two (on, off) for each relay channel

[image]

havanakuba, 2017-02-21:
Here are the scripts I created

I new to github, python and coding. I feel it should be done in the different way. Sorry for that. If someone can point me what I`m doing wrong, Really appreciate some help.

[scripts.zip](https://github.com/AstroPrint/AstroBox/files/791518/scripts.zip)

robermeyer, 2017-02-21:
Great work.
How do we get it into AstroPrint?

robermeyer, 2017-02-21:
@CoDanny CoDanny   
Is it possible for this to get moving some how?
I am confident many users would appreciate the feature.

havanakuba, 2017-02-21:
You encourage me to get this project moving, I did all of this today. :)

havanakuba, 2017-02-21:
now php script for all channels need to be write, the problem is I don`t know php :(

robermeyer, 2017-02-21:
It looks like the code is mostly JavaScript, so PH is not going to help.

CoDanny, 2017-02-22:
@robermeyer This isn't something that most of our users are even able to do. I agree though that some of our most advanced users will appreciate this. We can't add this functionality to the core platform but will certainly add it as a plugin when the plugin interface is ready

robermeyer, 2017-02-22:
What's the time frame for plugin interface?

CoDanny, 2017-02-22:
It's planned but we have other things to develop first. At this point I can't give you a date.

robermeyer, 2017-02-22:
That's not encouraging. In our engineering, planned without a date usually
means it not going to happen until somebody new comes in and changes
priorities.
Sounds like you are the guy the set priorities.

CoDanny, 2017-02-22:
@robermeyer Yes, I am (one of them). I'm the company's CTO. It doesn't mean it won't get done. Having a plugin API is important, we realize that it will let others contribute to the product and alleviate some of the problems and limitations that our most advanced users are facing. 

It's just that other things in the core product are currently in development. We're a small team and can't do everything at once. Note that the team not only works on the AstroBox but also on the AstroPrint Cloud (astroprint.com) and a few other companion apps to be released soon.

havanakuba, 2017-02-22:
Hi for now I`m using this temporary solution called WEBIOPi

https://github.com/doublebind/raspi

robermeyer, 2017-02-22:
For the cost of a Raspberry Pi and an AC SSR interface, its probably just a
DIY project to turn ON/OFF the printer separately.  It was nice having a
remote interface that would go through the firewall.
If I wait for this PlugIN to develop I will probably be working on my 4D
printer by then and won't need it anymore.  I have 2 printers so I will
need 2 Raspberry PI's anyway.  Whats one more :-)

I will look into the WEBIOPI myself.  Thanks for the tip.

havanakuba, 2017-02-23:
I have made custom interface to switch on/off light in my printer on raspberry pi 3 with installed astroprint on it.
install python and gpio

sudo apt-get install python-dev
sudo apt-get install python-pip3
sudo apt-get install python-rpi.gpio

then fallow https://github.com/doublebind/raspi

then 
sudo nano /etc/webiopi/config
in [HTTP]
change port to 8080 this step it is not essential will work without change.
uncomment and change path in
doc-root = /home/pi/www/

make www directory   in /home /pi

using filezilla copy index.html to /home/pi/www
[ch1.zip](https://github.com/AstroPrint/AstroBox/files/795422/ch1.zip)
[ch4.zip](https://github.com/AstroPrint/AstroBox/files/795424/ch4.zip)

sudo /etc/init.d/webiopi start - to start
sudo /etc/init.d/webiopi stop - to stop
sudo /etc/init.d/webiopi status - to check status

sudo update-rc.d webiopi defaults - add to autostart

havanakuba, 2017-02-23:
[image]
[image]

havanakuba, 2017-02-23:
[image]
[image]

Austinthemighty, 2017-02-23:
nice, currently i am working on a heated encloser.  In order to get the current info from astroprint i am using lxml to pull the data from the web page

sabbiolino80, 2017-04-09:
What about a custom gcode to switch the relays, with a parameter to select which one and a second parameter to specify on or off

sabbiolino80, 2017-04-09:
Similar to M107 to switch the fan

MikeDabrowski, 2017-04-10:
Alright guys take a look (here)[https://github.com/foosel/OctoPrint/wiki/Controlling-a-relay-board-from-your-RPi]. Changes are done to config.yaml. I did that on astro but it is not enough - the buttons does not show in right menu. So I went to /AstroBox/src/astroprint/templates/app.jinja2 (no idea what language is that - ruby?) and duplicated buttons used for shutdown and reboot. Now I think js is controlling that buttons so I'm gonna go through app.js (16k lines... yahhh) unless someone does have better idea. 

Probably we could use that scripts @havanakuba wrote. I think php can't be directly inserted into that jinja files. Also I don't know if js alone can manipulate gpio. Easies way would be to invoke `gpio mode 7 out` somehow.

I used octoprint set up as in that link I gave here but I had issues in maintaining connection to printer itself. (relay works very nicely, I'm using 1ch one). So far astro is maintaining connection. Hope we will come up with something together.

MikeDabrowski, 2017-04-10:
OK its in js 

```
doTurnoff: function() {
            $.ajax({
                url: API_BASEURL + "system",
                type: "POST",
                data: {
                    action: "shutdown"
                },
                success: _.bind(function() {
                    setTimeout(_.bind(function() {
                        this.$el.addClass("done");
                        this.$el.find(".icon-off").removeClass("blink-animation")
                    }, this), 6E3)
                }, this),
                error: _.bind(function() {
                    this.$el.find(".icon-off").removeClass("blink-animation");
                    noty({
                        text: "There was an error starting turn off sequence.",
                        timeout: 5E3
                    })
                }, this)
            })
        }
```
Does it mean that js can control system?

robermeyer, 2017-04-11:
Nice research MikeDabrowski

MikeDabrowski, 2017-04-11:
Ok, I fetched source code and looked it through. Good news its rather python doing system commands than js. Bad news I have no experience in python for web and have no idea where to place relevant command.

So if any of the creators could drop some hints which file to edit to place python code or system instructions we'd be very glad. Its very small change actually.  @CoDanny ?

Austinthemighty, 2017-04-11:
for python web programming it's basically HTML programming wrapped in python code

MikeDabrowski, 2017-04-11:
Yea, but which file to edit and how to make html button perform system level action?

Austinthemighty, 2017-04-13:
take a look at this https://bitbucket.org/fotosyn/fotosynlabs/raw/9d78640ca4ef8ffcea12194b8528b52ccad8b426/BerryCam/berryCam.py
it may help, but not by much
and this 
http://web2py.com/
http://web2py.com/books/default/chapter/29/13#One-step-production-deployment
there
https://www.instructables.com/id/Simple-and-intuitive-web-interface-for-your-Raspbe/

i havent done any web python stuff but here is some stuff on it

robermeyer, 2017-04-21:
This is perfect.
I just need one 3.3Volt IO to turn power on and off.  $19.99 kit.  
http://www.powerswitchtail.com/Pages/PSTKKit.aspx
$25.99 assembled
http://www.powerswitchtail.com/Pages/default.aspx
[image]

With Octoprint supporting it already, this would seem very trivial for one of the AstroPrint Gurus.  
With the adaptor available its a no brainer.

MikeDabrowski, 2017-04-21:
@robermeyer Man its an overkill, simple relay module is for 1.3USD approx on ebay, works great just pay attention if its 3.3 or 5v. Problem is in astrobox not supporting custom switches. I have no time to dig deeper and other issues preventing me from using astrobox. But maybe during summer I will try to do something about it

robermeyer, 2017-04-21:
I am looking for something with no exposed 110Volts.
Simply connecting 2 wires to the RASpi and your ready to go.  Safe and
reliable.
Those jumper wires on that $1.30 board don't work for me.
The Youtube videos show how unreliable those jumpers are with having to
debug such a trivial interface.
Those $1.30 relay boards don't have a grounded plug or socket or protective
case.
That is fine for proof of concept.  I already know it works.  I want a
production version 🙂
You have to have the Plug, Socket and case to make it useful and not
dangerous.
I have kids that like to touch shiny stuff!
$20.00 is cheap if AstroPrint had 10-20 lines of code to turn it ON/OFF.
I just bought another RASPi to install OCTOPRINT.  I will see how well this
works with this power box.
AstroBox may be better in some ways but in others they are moving very slow
right now.

kodacy, 2017-04-26:
Fell the same need, and agree that this option should be made available has soon has possible, since it is such a simple procedure that can have so many applications (energy save, light and appliances related to the print control).

raptordemon, 2018-01-01:
Hi has any progress been made on this?

CoDanny, 2018-01-02:
@raptordemon No, at the moment we're busy with other things. When the AstroBox API is released this will be possible by the community

loadedsith, 2020-02-27:
PSU toggling support is one of 2 plugins that made me make the conversion back to OctoPi. I miss my Astrobox, please support this sort of plugin.

robermeyer, 2020-05-24:
Unbelievable that this feature request went in Mar 2016 and its still not implemented.
I finally converted all my boxes over to Octoprint with OctoClient on my phone with WiFi switch to turn printer and lights on/off and it works great.
I use symplify3d to slice.  Octoprint works with my camera.  AstroPrint stopped working with same camera some time back.

jlopfri921, 2022-01-04:
> PSU toggling support is one of 2 plugins that made me make the conversion back to OctoPi. I miss my Astrobox, please support this sort of plugin.

[Post AstroBox GPIO turn on/off printer 3D](https://forum.astroprint.com/t/turn-on-off-remote-printer-and-lights-relays-contactor-tutorial/6476)