AstroPrint/AstroBox #91: Feature Request: Chain Gcode Commands (separated by ";" )

filipgoc, 2016-03-25:
It would be lovely if the gcode terminal automatically parsed commands separated by semicolons!
(Like marlin does internally.)

For example, imagine this gcode chain for calibration purposes: 

### instead of
```
G28 (send)
G29 (send)
G1 X50 Y50 F5000 (send)
```

### I'd do 
```
G28; G29; G1 X50 Y50 F5000 (send)
```

avluis, 2016-03-25:
That would definitely be lovely~

CoDanny, 2016-03-25:
I thought that anything that follows a ';' is ignored as a comment.

filipgoc, 2016-03-28:
@CoDanny of course! 
I can't even believe what I had said. I got confused with another interface.

MARLIN 
has a function for parsing commands into new lines ... through deliberate \n

```like so (paraphrased):
enqueuecommands("G00\nG00\nG00")```
 
*What about deliberate support for "\n" then? Sounds reasonable!*
Even typing/pasting multiple lines of codes would in a way work.