AstroPrint/AstroBox #97: lost connection

Promic2, 2016-04-20:
When im done with a print the astro disconnect from printer
an can not connect again
i have to Restart the Rasberry pi ( Power off ) 
an let it boot up and i works for 1 print again 

when only restarting the printer ( power off ) 
it´s the same issue . cant connect to printer

CoDanny, 2016-04-20:
What version of the astrobox are you using?

Promic2, 2016-04-20:
just updated to i think is 8(7) 
im printing right now .. so i cant see the settings page

CoDanny, 2016-04-20:
Ok, that version should have a fix for a similar problem. What printer?

Promic2, 2016-04-20:
i have no problem when connected directly to pc

CoDanny, 2016-04-20:
We test here with a Wanhao i3 duplicator with no problems. Do you still have the problem on 0.8(7) If you do, please let's move this over to the a [support ticket](https://astroprint.zendesk.com) and we'll open an issue here if something is found to be fixed

Promic2, 2016-04-20:
yes the problem came with this update  0.8(7) .
but add a version number at the top of astrobox .. for when printing the settings is lock

CoDanny, 2016-04-20:
ok, please open a support ticket we'll continue this there. I'm going to close this for now

closed 2016-04-20 by CoDanny