AstroPrint/AstroBox #124: Camera Settings Not Displaying 0.9(1)

jneilliii, 2016-06-18:
When going into settings to set encoding for the video stream to support chrome I get a message about the screen resolution being adjusted at the top in an orange bar and then the page never shows any settings to change.  I was able at one point to get the settings to show (don't know how) and changed the encoding and when saving get an error about the resolution not being supported, but can't change the resolution.

CoDanny, 2016-06-18:
@jneilliii What camera do you have?

jneilliii, 2016-06-18:
It's a gear head usb camera. I know it works with Linux as it was fully functional on my octoprint install.

CoDanny, 2016-06-18:
@jneilliii thanks. 

Note that OctoPrint doesn't do proper video. They do a rapid succession of JPEG images in an HTML image tag which gives the (very good) impression of a video.

Having said that, I'm pretty sure your camera is supported but we might have missed something. Could you make your problem happen again and immediately after send us logs via the advanced settings section?

This will help us with our investigation.

jneilliii, 2016-06-18:
As soon as my next print is finished I will do exactly that.  FYI, the capturing of a single still image does not work either.

jneilliii, 2016-06-18:
Logs just sent referencing this issue number.

CoDanny, 2016-06-20:
Could you execute this on your pi with the camera connected?

`v4l2-ctl --list-formats-ext`

jneilliii, 2016-06-20:
ioctl: VIDIOC_ENUM_FMT

        Index       : 0
        Type        : Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name        : Motion-JPEG
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1024x576
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1392x768
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1600x896
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name        : Motion-JPEG
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1024x576
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1392x768
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1600x896
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)

CoDanny, 2016-06-20:
was that it? no other formats?

jneilliii, 2016-06-21:
That's definitely it, here's the file.
[formats.txt](https://github.com/AstroPrint/AstroBox/files/326784/formats.txt)

CoDanny, 2016-06-22:
Thanks @jneilliii. Could you run this too so that I know exactly which camera you have?

`v4l2-ctl -D`

jneilliii, 2016-06-23:
```
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : FULL HD 1080P Webcam
        Bus info      : usb-3f980000.usb-1.2
        Driver version: 4.4.11
        Capabilities  : 0x84200001
                Video Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
```

commit a85aa9b8 (2016-06-23): fixes #124
closed 2016-06-24 by CoDanny