AstroPrint/AstroBox #149: Remove printing message

FernandoGarcia, 2016-09-23:
Hi!

Please remove the "printing..." message from info screen after cancel a print.

Best regards.

CoDanny, 2016-09-23:
Could you provide a screenshot of the message that you're referring to?

FernandoGarcia, 2016-09-23:
Hi!

This is the default status:

[image]

This is the status after cancel a print:

[image]

Here you can see the "Printing..." message.

FernandoGarcia, 2016-09-26:
Hi!

This message remains on LCD when the print is finished too.

When printing from SD card the message is like this:

[image]

Best regards.

robotdanb, 2016-12-13:
I apologize if this seems out of the blue, or if the topic is a little outdated, as this is my first post here, but I believe what might be causing that message is the firmware of the printer itself. I've run into similar behavior with mine before, and it's technically not related to Astroprint or the GCODE itself.

Short answer : Try adding the line "M117 Print Cancelled." (without quotes) to your "Cancel GCODE" field under what you have already there, if applicable.

Long answer: The Marlin firmware (I'll admit it's an assumption on my part that's what your printer uses) uses the Gcode "M117" to display a message. Typically, many slicers don't use it much during printing itself. When cancelling a print, Astroprint will stop sending the pre-written code, so whatever messages would appear down the line wouldn't be sent. Marlin-based printers seem to have default messages that are displayed during certain events (such as during preheating or homing operations). Without being overwritten through Gcode, the message "Printing..." may be displayed by default while the printer is receiving Gcode commands (including the cancel commands), and doesn't change the message unless told to do so (hence it still displaying). Adding the M117 code shown in the short answer will manually send the message to force a change.

In addition: Should the code work for you, you can also add a message override when a print is complete in the same way.

Mild disclaimer: As I'm still learning the intricacies of Gcode, there is a slight chance that this may not work as intended. I have tested this code on my printer, and it works without issue. I have an Anet A8 (Prusa i3 clone using Marlin).

Again, I do apologize if it seems I'm barging in out of nowhere and throwing around assumptions...lol

FernandoGarcia, 2016-12-13:
Hi!
I don't think that edit the Gcode manually is good idea. It should be automatic as in Repetier host.
After open this issues I did some upgrades in Astroprint and now it's worse because I can't see even "printing..." message. After start the print I can see "Heating..." message all time.
Best regards.

FernandoGarcia, 2018-05-27:
Hi!
I'm back to Astroprint since now has support to dual extruder.
I'm using 0.12.5 an the LCD is not showing the print time or remaining time during a print.
I shows only "heating done".
Best regards.

FernandoGarcia, 2018-05-28:
[image]