AstroPrint/AstroBox #150: Strange behaviour

FernandoGarcia, 2016-09-23:
Hi!
I was starting a print to take some images to answer your question in other issue and noticed a very strange behaviour.

After start the print via file manager the Astoprint sent the command to heat the bed as you can in the image:

[image]

But the monitor is showing the target temperature to bed as zero and nozzle as 240°C:

[image]

In this moment the monitor should display the target temperature to bed as 85°C and nozzle as zero.

After the bed heating is done a command to turn off the bed was sent but it's wrong the temperature should  remain as 85°C and 240°C for nozzle.

[image]

The strange thing is the monitor display the right temperature to both but the info on LCD is totally different.

[image]

The print was started with the bed OFF:

[image]

I'm using a Geeetech Prusa I3 Pro B with Marlin.

I did the upgrade to version 0.9(5) yesterday, I didn't see it before. Moreover I'm new to Astroprint.

Best regards.

CoDanny, 2017-04-27:
Does this still happen with 0.10(7)?

FernandoGarcia, 2017-04-27:
Since I have installed a second extruder I'm not using Astroprint because the support is very poor.

I'm waiting for updates.

CoDanny, 2017-04-27:
OK, I'll close this now. Support is coming :)

closed 2017-04-27 by CoDanny