AstroPrint/AstroBox #161: Random Time shows up in time left section at end of print

joshwhite47, 2016-10-18:
On the last layer with a few seconds left a random time(or what seemed like a random time) showed up in the time left area. The print goes down to 0 switches to this time and then the print ends.

[image]

joshwhite47, 2016-10-18:
Here is a video where this issue happens. 

https://drive.google.com/file/d/0BxCcy5TynrdaQlZwUXFBSEJ0Yk0/view?usp=sharing

lukemaster, 2016-10-19:
In the first capture of this issue, I can see 19:38 minutes of time left. Is this the complete time left of this printing? When the progress is 0 percent, how much time left appears in Time?

CoDanny, 2016-10-19:
Fixed on astroprint.com

closed 2016-10-19 by CoDanny