AstroPrint/AstroBox #185: Printer not connecting on power up of astrobox

joshwhite47, 2016-11-22:
If I turn on a box and then see that box on astroprint.com, AP desktop, or AP mobile I see the box connected to my astroprint account but I can't print because the printer is not connected to the box. As soon as I open the local page on the astrobox the page loads and the printer connects on page load.

CoDanny, 2016-11-22:
This has always been like this. However it might make sense that the printer connects on startup. That is a new feature.

joshwhite47, 2016-11-22:
Yeah, I guess I've just noticed it more lately since i've been trying to look at printers from the app, and desktop software more lately. It would be nice for it to detect a connected printer without having to open that page.

commit 78d2dd8c (2017-02-06): fixes #185
closed 2017-02-06 by CoDanny