AstroPrint/AstroBox #194: OrangePi version or manual linux install

pabloevaristo, 2016-12-09:
I have been using Astroprint services for a while. And now I would like to use it as a standalone printing device, but I have a orangepi PC Plus. A great mini computer. 
It can run raspbian flawlessly, so i suppose that it would not be hard to translate the astroprint code to this system... Is that possible?
Another option would be to have instructions on how to install manually astroprint on a linux system so orangepi owners could set astroprint up on these boards.
THANKS

CoDanny, 2016-12-09:
Unfortunately we don't have the resources to support any more platforms. The project is open source and you're welcome to port it on own.

Sorry but we just need to focus on one platform so that we can go faster and offer better service.

closed 2016-12-09 by CoDanny

pabloevaristo, 2016-12-09:
And how about manual installation steps for raspbian?
I suppose you are using that distro as foundation of astrobox
implementation