AstroPrint/AstroBox #202: Display temperarure for second extruder

FernandoGarcia, 2016-12-13:
Hi!

Please add a second control bar for second extruder.

As it's now is impossible know the temperature for both extruders.

Now I have the first extruder with 66°C and the second with 154°C but I can see the temperature only for the first extruder.

[image]

Best regards.

JulianGarciaCastillo, 2016-12-13:
It's a feature we are  currently working. Support for multiple extruders. It will be added soon.

highground88, 2017-10-17:
I notice this comment was a while ago, Astroprint still broken apparently in this regard on the latest current version on my Flashforge Creator Pro - or is it just me?

CoDanny, 2017-10-17:
@highground88 Not just you. We have active branch for this

closed 2018-04-08 by CoDanny

FernandoGarcia, 2018-04-08:
Thanks for this new feature!
I'm back to Astroprint. :)