AstroPrint/AstroBox #215: Cancel doesn't stop startup-preheating

wwendorf, 2017-01-24:
I have hit cancel before the print actually begins, during the heatup phase.  When I do this, the print cancels just fine, however, the heatup continues until bed temp and hotend temp have been reached, then it will finally shut off.

The only way I have to cancel the heatup from the aborted print is to reset the actual printer itself.

Restarting astrobox does not reset the heatup routines.

CoDanny, 2017-01-24:
This is unfortunately how the firmware works :( Once the hardware is issued a heat and wait command, it will not receive any other command until the printer has reached the temperature. We send the commands as soon as you press the cancel button but the printer just queues them

closed 2017-01-24 by CoDanny

wwendorf, 2017-01-24:
Well, that is strange, because 10(3) when I hit cancel, I never had that
problem?  If I hit cancel, everything canceled.

Wade

CoDanny, 2017-01-24:
@wwendorf The UI was misleading in previous releases. It seems like it cancelled but the printer was still heating and if you had tried to control the printer, the commands would've been queue'd

wwendorf, 2017-01-24:
OK, thanks for the input.