AstroPrint/AstroBox #217: Filament sensor

Austinthemighty, 2017-01-30:
Can you guys add a filament sensor.  I am thinking the main idea would be that the filament would run over a [image] and close the circuit when there is filament and would open the circuit when there is no more filament and stop the print.

beachmiles, 2017-04-14:
Yes please!!

databloom, 2018-06-11:
Actually, I built the simplest possible airprint detector today for my dual extruder makerbot. I took an optical usb mouse and put it beside my filament reel, as the reel moves, I get movement in my evdev events in python. That's it. I'm not joking, I can detect filament out and airprint in immediately. Let me know if you guys want to see my tragic code and pull it in to Astroprint somehow.

loadedsith, 2020-02-27:
Filament runout sensor is one of 2 plugins that made me make the conversion back to OctoPi. I miss my Astrobox, please support this sort of plugin.

KyleBoyer, 2020-06-26:
What would it take to support the EZOut sensor from TH3DStudio?