AstroPrint/AstroBox #229: Local API key gets reset on every reboot

cbgj, 2017-02-07:
The "octoprint" API key found in the advanced settings menu is reset on every reboot. This occurs both on ungraceful and graceful shutdowns/reboots.

CoDanny, 2017-02-07:
You can lock it by editing your /etc/astrobox/config.yaml file and adding:

```
api:
  key: whateveryouwant
```

cbgj, 2017-02-07:
I still consider this a feature request then. Its not very intuitive. Considering the plug n play nature of this software. You should not have to tinker with config files when there is a good UI.

CoDanny, 2017-02-07:
@cbgj I agree with the simplicity. However fixing the API Key is hardly a novice user action. Help me understand two things:

a) Why do you need to fix it? Are you aware of the security implications?
b) Are you able to actually change the config file? if so, would a user with your profile be able to do it pretty much always.

My point being simply: this doesn't seem to be a regular user feature and thus is somewhat hidden in purpose.

Of course I can always be convinced otherwise :)

cbgj, 2017-02-07:
a) Yes, a user must actively choose to open ports to the outside their NAT'ed network for this to be a security risk. By this logic the default username and password for SSH should be changed as well, since someone else on the network might be snooping around (actually thats a good idea; to set the SSH password in the first time setup for security hardening).

b) Yes, i could fix it myself. I just think this is a feature that should be plug and play. And it is also the default behavior in OctoPrint last time i used it which is why i was surprised that it was not in AstroPrint.

c) You should rather be promoting this feature to show how easy the workflow can be with external slicers with the right tools and integrations. Since many people proabably are using the slicer they are used to before coming to AstroPrint, with profiles for different filaments, etc, etc.

cbgj, 2017-02-15:
@CoDanny have you considered my points above?

Austinthemighty, 2017-02-15:
also the username and password for the pi should be the username that is set up on the astroprint

CoDanny, 2017-02-16:
@cbgj Yes, I'll schedule locking the API Key for the next release. Would that be ok for you? We won't actually let you set the key but we'll allow you to set lock it so that it doesn't change on reboot.

CoDanny, 2017-02-16:
@Austinthemighty Let's open a new issue for that. It's hard for us to track to different features/suggestion in the same issue ticket.

cbgj, 2017-02-16:
@CoDanny: Yes, that would make us both happy. Great solution!

commit bd210c9a (2017-04-21): closes #229
closed 2017-04-21 by CoDanny