AstroPrint/AstroBox #234: Username and password

Austinthemighty, 2017-02-17:
the username and password for the pi should be the username that is set up on the astroprint server/login

CoDanny, 2017-02-20:
astroprint username is an email. So far we have taken the policy of separating the Raspbian Operating System from the AstroBox software. With this in mind there are two cases:

1) You're an advanced user and like to go into the command line. In this case you should set your own password and username to enter the OS.

2) You're not knowledgeable in Linux and just want to use the AstroBox as your Printer Controller. In this case you might never have to go into the command line and thus pi username/password is irrelevant.

Austinthemighty, 2017-02-20:
But if I am not an advanced user and I open a port to my pi I now have a hole in my security

CoDanny, 2017-02-20:
I would consider that pretty advanced. If you do that, then you should definitely at a minimum change the default pi password. Also note that with AstroPrint that is not needed.

Austinthemighty, 2017-02-21:
most users probably don't know how to change the password, or even know that the password is set to that, also i thought that the password was set to something else for AstroPrint.  also i port forwared at age 10 with out knowing what i was doing, thanks to youtube videos.