AstroPrint/AstroBox #248: Control screen doesn't live update since v0.10(7)

PxT, 2017-03-29:
With the last update the Controls screen doesn't live update the extruder temperature any more.  Temp always just displays as "--" even though the printer is responding normally.  M105 entered into the terminal responds as expected.

Debug serial.log has contents like this:
>2017-03-29 02:31:07,746 - SERIAL - DEBUG - Enabling serial logging
>2017-03-29 02:31:21,004 - SERIAL - DEBUG - Send: M104 S189.000000  
>2017-03-29 02:31:21,013 - SERIAL - DEBUG - Recv: ok  
>2017-03-29 02:33:19,201 - SERIAL - DEBUG - Send: M104 S159.000000  
>2017-03-29 02:33:19,211 - SERIAL - DEBUG - Recv: ok  

Printrbot connected to astro box on a Raspberry Pi 2 Model B.  Let me know if you need any other info.

CoDanny, 2017-03-29:
Ok, does it start showing data if you modify the temp using the slider? Is the temperature reported when printing?

PxT, 2017-03-29:
When printing it shows normally.  But from the control screen it is always "--" regardless of the slider - the DEBUG logs above are me moving the slider up and down.

CoDanny, 2017-03-29:
The debug command don't show the M105 commands which requests the printer for temp updates. Is the GCODE Terminal open?

PxT, 2017-03-29:
Not sure what you mean, is it open at what point?  When I'm on the control screen the terminal is not open -- I'm not sure that's possible unless I open a second browser window, right?

CoDanny, 2017-04-27:
I meant exactly that, whether you had the GCODE terminal open in another windows.

I'm not able to reproduce this. Does it happen for you all the time?

PxT, 2017-04-27:
It happened 100% of the time on my setup, however I've switched back to Octoprint since writing this bug because it has some other features that I'm taking advantage of, so I don't have a way to reproduce it at the moment.

If you don't see the issue locally then OK to close this I guess.

closed 2017-04-27 by CoDanny