AstroPrint/AstroBox #269: Hot end stays on model after print finishes

olskar, 2017-05-05:
Hot end stays on model after print finishes. I am using latest version. Expected behavior is homing all axes after print finishes.

CoDanny, 2017-05-05:
What do you use to create your print file? If AstroPrint Slicer, is it a custom printer? Check your End GCODE...

olskar, 2017-05-05:
Yes, I am using Astroprint slicer with Anet A8, there is a profile for it in astroprint. I managed to make it home by adding "G28 X0 Y0 ;" under advanced settings. However as I understand it this project is mostly towards not very techinical people (compared to octaprint for example) so it would be nice if it could work out of the box

CoDanny, 2017-05-05:
You're correct. It's mistake on the Anet profiles which have been corrected.

closed 2017-05-05 by CoDanny