AstroPrint/AstroBox #304: android 8.1 cant login

rhys100, 2017-10-29:
Using a pixel XL, when i log in I get the message

"There was an error"

"We encoutner an error performating the operation: {{info}} "

CoDanny, 2017-10-30:
@rhys100 please don't report mobile app errors here. Use the feedback link in the app settings

closed 2017-10-30 by CoDanny

rhys100, 2017-10-30:
One cannot submit an error if they are not logged in on the app

CoDanny, 2017-10-30:
Good point. Still try other means for the future, like opening a support ticket here: https://astroprint.zendesk.com/hc/en-us/requests/new

This repo is only for the AstroBox open source project. 

**Regarding this problem**: The issue is likely that you had a beta version installed. In the past the 
format we used to store data changed. Try completely uninstalling the app before installing the version on the store.