AstroPrint/AstroBox #321: Plugins

forum999, 2017-12-18:
Enable plugins like Octoprint to allow for filament sensors

or ideally save u the work and allow octoprint compatability o use their plugins to save the work of writing new script... Astroprint severly lacks in this area

CoDanny, 2017-12-18:
We will offer this in the future. Also this: https://github.com/AstroPrint/OctoPrint-AstroPrint

forum999, 2017-12-19:
hi thats great but i am a diehard astroprint user and often promote it over octiprint which i am not keen on at all, i would much prefer the ability to create these scripts or impliment existing ones directly into astroprint , especially the filament sensor

CoDanny, 2017-12-19:
Thanks :) We will release a plugin API early in 2018

closed 2017-12-19 by CoDanny