AstroPrint/AstroBox #325: Investigate using the internal RPi3 WiFi as concurrent AP, Client Mode

CoDanny, 2017-12-22:
More info here: https://albeec13.github.io/2017/09/26/raspberry-pi-zero-w-simultaneous-ap-and-managed-mode-wifi/

Rigid3D, 2017-12-22:
This method works very well on Octoprint/Pi Zero W. Tested on Raspberry Pi Zero W. 

But it was not succesful on Astroprint/Pi 3. I couldn't manage to login system without a wifi dongle on pi Zero W. :) and tested with Pi 3.

CoDanny, 2018-02-23:
Another (easier?) way to get the second interface up:
https://www.raspberrypi.org/forums/viewtopic.php?t=138730#p936518

Still need to make changes in AP software to use it.

CoDanny, 2018-02-27:
I'm unable to connect to an external WiFi while connected to the AP. This is likely not going to work for us. It appears that for this to work the external WiFi settings needs to be hardcoded in the /etc/interfaces which won't work for us.

closed 2018-02-27 by CoDanny