AstroPrint/AstroBox #384: Tevo Tornado prints getting stopped automatically after couple layers

suvinks, 2018-04-03:
Hi 

Am using Tevo tornado firmware 1.1. I am able to connect and monitor the prints. However, for a print more than couple layers, the print gets paused automatically after 2-3 layers. 
The printer controls still work if I cancel the print and move the motors. However, due to some reason it will not print.

Any suggestions?

I was printing astrobot keychain file which was sliced in astrocloud. I also printed a very small 2layer print which came out correctly. Anything more than 5 has issues.

Thanks
Suvin

CoDanny, 2018-04-03:
Activate serial logs and when this happens again send to us via Settings -> Sofware -> advanced

suvinks, 2018-04-04:
I have sent the log over with ticket number 384

suvinks, 2018-04-04:
2018-04-04 02:32:10,626 - SERIAL - DEBUG - Send: N10544 G1 F1800 X159.074 Y161.150 E858.59670*13
2018-04-04 02:32:10,652 - SERIAL - DEBUG - Recv: ok
2018-04-04 02:32:10,653 - SERIAL - DEBUG - Send: N10545 G0 F9000 X158.466 Y161.193*71
2018-04-04 02:32:12,657 - SERIAL - DEBUG - Read communication timeout during printing, listen again
2018-04-04 02:32:22,670 - SERIAL - DEBUG - Read communication timeout during printing, listen again
2018-04-04 02:32:32,682 - SERIAL - DEBUG - Read communication timeout during printing, listen again
2018-04-04 02:32:42,695 - SERIAL - DEBUG - Read communication timeout during printing, listen again
2018-04-04 02:32:52,708 - SERIAL - DEBUG - Read communication timeout during printing, listen again
2018-04-04 02:32:54,703 - SERIAL - DEBUG - Send: G28 X0 Y0
2018-04-04 02:32:54,705 - SERIAL - DEBUG - Send: N0 M110 N0*125
2018-04-04 02:32:54,706 - SERIAL - DEBUG - Send: M140 S0.000000
2018-04-04 02:32:54,707 - SERIAL - DEBUG - Send: M104 S0.000000
2018-04-04 02:32:54,707 - SERIAL - DEBUG - Send: M84
2018-04-04 02:32:54,708 - SERIAL - DEBUG - Send: M106 S0
2018-04-04 02:32:54,794 - SERIAL - DEBUG - Changing monitoring state from 'Printing' to 'Operational'
2018-04-04 02:32:56,714 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:00,720 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:06,728 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:12,737 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:18,751 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:24,760 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:30,772 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:36,783 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:42,791 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:44,020 - SERIAL - DEBUG - Changing monitoring state from 'Operational' to 'Closed'
2018-04-04 02:33:44,035 - SERIAL - DEBUG - Changing monitoring state from 'Offline' to 'Opening serial port'
2018-04-04 02:33:44,049 - SERIAL - DEBUG - Connecting to: /dev/ttyUSB0
2018-04-04 02:33:44,056 - SERIAL - DEBUG - Connected to: Serial<id=0x73e47e50, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=True, rtscts=False, dsrdtr=False), starting monitor
2018-04-04 02:33:44,057 - SERIAL - DEBUG - Changing monitoring state from 'Opening serial port' to 'Connecting'
2018-04-04 02:33:44,058 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:44,795 - SERIAL - DEBUG - Connection closed, closing down monitor
2018-04-04 02:33:46,061 - SERIAL - DEBUG - Send: M105
2018-04-04 02:33:48,065 - SERIAL - DEBUG - Send: M105

CoDanny, 2018-04-04:
This log tells me that the printer stopped receiving commands or sending responses. After this:

```
2018-04-04 02:32:10,652 - SERIAL - DEBUG - Recv: ok
2018-04-04 02:32:10,653 - SERIAL - DEBUG - Send: N10545 G0 F9000 X158.466 Y161.193*71
2018-04-04 02:32:12,657 - SERIAL - DEBUG - Read communication timeout during printing, listen again
```

There's nothing else coming form the printer. I would guess it's a firmware or board issue.

suvinks, 2018-04-04:
I connected the printer directly to my PC and used Ultimaker Cura to print directly. It did so without any issue. If it was a firmware issue the usb communication should have stopped for cura too right?
Also my tornado is having Marilin 1.1 firmware. Which one do you suggest?

CoDanny, 2018-04-04:
The firmware is fine but the loss of communication I can't explain. We have a Tevo Tarantula that we showing the same problems. Tevo sent us a new board, we're going to be testing it this week

suvinks, 2018-04-05:
Thanks. Do let me know how it goes. I did update the firmware to 1.1.6 but still have the communication issue. This one too is printing with Cura without issues. With Astroprint, it stops after 3 layers :(
I did try your Astroprint desktop. That one started printing without even heating up the bed or nozzle or even homing. I had to cancel the print. Looks like there is an issue with Gcode creation on that one. I could have tested the usb printing on that if it was working correctly.

suvinks, 2018-04-13:
Is your Tevo working now without issues?

CoDanny, 2018-04-13:
We haven't had time to test yet

DanielJoyce, 2019-06-05:
Your raspberry pi needs a good powersupply and your printer and pi need to share a ground, like plug them both into the same outlet or power strip. This smells like brownout issues, especially if it worked from a laptop.

closed 2020-03-11 by CoDanny