AstroPrint/AstroBox #428: Skip password or store password on Pi

urbanxl, 2018-09-23:
Everytime Astroprint start a password is requested. It is not possible to write password with the touchscreen without connecting a keyboard, and there is no remember password box.

So, is it possible to load the page local without password protection or at least be able to save password?

urbanxl, 2018-09-25:
Hello,
I'm using my own raspberry Pi with a touch screen running Astroprint into Chromium in full screen. So, I'm not using the Astrobox sold by Astroprint. I can write the password (for astroprint Cloud) with the keyboard but I want to avoid this manual task each time I start Astroprint. I would excepect to store the password, or something else, avoiding this "login" page on start. I don't find a way to do that.

CoDanny, 2018-09-25:
I see. Using the software in that way is unsupported. Does't mean you can't use it like that but we don't get involved in supporting use cases for these types of configuration. I would look into installing an on-screen touch keyboard

closed 2018-09-25 by CoDanny

urbanxl, 2018-09-25:
Sorry to make you unhappy. I have understood that your business was to win money with the cloud. If because of the cloud I have a bad experiences (=log on every start) I will stop to used the cloud and my issue will be solved. Also, I can switching to another solution too. I don't really understand your concern because all the tips and support to have this kind of configuration have been found on Astroprint forum with answer from Astroprint team.

Maybe I don't understand why I'm using the software in an "unsupported" way. Does it is because of a security risk? if, I'm out of the spirit of Astroprint and/or GNU I stop my use case. Ask me and I do it.

Meanwhile, except log on each start everything is working very well and it looks like very nice from my point of view.

CoDanny, 2018-09-25:
I'm not unhappy at all :)

It's simply unsupported by us simply because we don't have the resources to focus on all the ways our software can be used. We don't mind you doing it at.

urbanxl, 2018-09-25:
OK. Thank you.
Sorry, I'm french and "unsupported" could have 2 meanings.

So I understand that my ask is not a "bug" but could be an "improvement" that you are not able to tackle because of resource. It is an important information for me.
I think it's too bad for the project to close the ticket, even if you are not able to work on it. It should be part of the "backlog".

And, again, it works very well ! You are doing a good job !

Thank you.