AstroPrint/AstroBox #430: GPIO TTL SERIAL - No Connection to Printer

Rigid3D, 2018-09-29:
With the latest version of Astrobox 0.13(16), printer is not detected and can not connect to the printer via pin 14 and 15 on GPIO. Port is not detected.

It works flawlessly on Astrobox 0.10(9).

Board is Raspberry Pi 3 and printer firmware is Marlin 1.1.9

There is no record on serial log.

Astrobox.log related record is

astroprint.printer - WARNING - Unable to connect to any available port [] with baudrate [115200]

CoDanny, 2018-10-01:
I think I have the fix. Could you please run these commands in your box to verify it?

```
# sudo wget https://raw.githubusercontent.com/AstroPrint/AstroBox/fix-430/src/astroprint/printer/marlin/__init__.py /AstroBox/src/astroprint/printer/marlin/__init__.py
# sudo service astrobox restart
```

Then see if it works.

CoDanny, 2018-10-10:
@Rigid3D Were you able to verify that my changes helped you?

CoDanny, 2018-12-12:
@Rigid3D I wanted to see if we should merge this in the codeline. Were you able to verify that it works?

commit 9e0fa6f3 (2018-10-01): fixes #430
commit 49c11029 (2019-01-10): Merge pull request #473 from AstroPrint/fix-430

fixes #430
closed 2019-01-10 by CoDanny