AstroPrint/AstroBox #438: Print Queue: Canceling a print job shouldn't remove the file from the queue

CoDanny, 2018-10-22:
Canceling a print job should leave the item in the queue, in case the user wants to retry

julianCast, 2018-10-23:
Canceling a print job just change its status to canceled or failed, but it doesn't remove it from the queue, just appear in Finished section.

Could it be a print job that was printed directly, without being in the queue? In that case,  the print job disappear from the queue because it was never part of it.

CoDanny, 2018-10-23:
I'm pretty sure it was in the queue. I didn't look in finished but I think that perhaps is more intuitive if only successfully finished jobs go to the finished tab. When a job is cancelled or failed it might be better that it stays in the queue so it can be easily retried.

The user can always delete if they don't want to retry