AstroPrint/AstroBox #454: Allow Locking of API Key between reboots #229

supuflounder, 2018-12-08:
I have an AstroBox Touch.  I found this article, which tells me to modify a file in /etc.  My laptop has no such file (MacOS X) and I thought that maybe the file was on the boot card for the AstroBox Touch, but I can find no sign of it.

What is it, why would I need it, and why isn't this just pure "plug and play"? I expect it to work "out of the box", but it doesn't.  My sclic3r app refuses to use it because there is no API key given, but since I have no idea what this is or how to find it, it seems a bit over-the-top to require it

CoDanny, 2018-12-08:
Where have you seen that you can use Slic3r with an AstroBox? You might be confused with OctoPrint...

closed 2018-12-08 by CoDanny

supuflounder, 2018-12-08:
I did it sucesfulky thrrr days ago, and there was a nonblank API key.  But
after restarting slic3r after a reboot, I went to print, and IoT had a
blank printer name and a blank API key.  So it has already worked.

   Joe