AstroPrint/AstroBox #540: Logitech C270: stream timeout

CoDanny, 2019-11-10:
I am using a Logitech C270 webcam with my AstroBox and have been having a recurring issue where I can never monitor my prints with the webcam because the webcam takes to long to power on when I press the view webcam feed button and AstroPrint times out before the camera starts to stream its footage.

I-like-stuff, 2020-07-06:
Has there been any progress on this as I too have the same issue, however I am sure on a version of AstroPrint more than a year ago I had it working. If you need any information I am able to provide.

CoDanny, 2020-07-06:
We needed to pick some cameras a several price ranges to ensure we test with them. Unfortunately this wasn't one fo them: https://astroprint.zendesk.com/hc/en-us/articles/203050005-What-cameras-can-I-connect-to-the-AstroBox-Touch-and-Gateway-

closed 2020-07-06 by CoDanny

I-like-stuff, 2020-07-06:
Why would this not work on AstroPrint when it does on octoprint, it is therefore Linux compatible.

CoDanny, 2020-07-06:
I can't really answer that. OctoPrint and the AstroBox are very different in the way the gather data from the camera and produce video. Unfortunately we can only test with a few models and while we know that many other do work also, we can't in any way test all cameras out there.