AstroPrint/AstroBox #564: Not receiving printing finished emails

jefft0825, 2020-05-11:
Logged into my account.   It’s setup to receive them but I haven’t been getting them for a few weeks now.   It might have been when I updated to the newest version but not sure.  Any suggestions?

CoDanny, 2020-05-11:
Do you see your completed print jobs in the cloud's print history section?

jefft0825, 2020-05-11:
Under the cloud section in file manager?     I have never been able to see
completed print history

jefft0825, 2020-05-11:
Just found what you were asking about.   There is some print history there
but it stopped on 4/29.

CoDanny, 2020-05-11:
I think this might be a problem related to you box and not a product issue. I'll close this issue as this is for product issues. Please open a support ticket with us so we can take a deeper look at this.

closed 2020-05-11 by CoDanny